martynwheeler/u-lora #6: Recipient address

sea-of-rob, 2022-07-13:
In the original docs for raspi-lora library they state that a message can be broadcast to all devices irrespective of their address:

`header_to` Address of recipient (0-255). If address is 255, the message will be broadcast to all devices

Is this supported as I cant get it to work with:

`lora.send("This is a test message", 255)`

Thanks

atonughosh, 2022-11-20:
I'm also having the same issue.
How to broadcast a message using this library?

barryhunter, 2023-10-05:
There is a bug in this library, kinda inheritied from raspi-lora. Even raspi-lora wouldnt receive broadcast messages, unless `receive_all` was used. This project tried to fix it, but not quite. 

I found and fixed this in my own fork of this project
https://github.com/barryhunter/u-lora/commit/685edbc60dfe9d3bff405ef586f71253269da6b8#diff-82727dcdad31bd5c8d91b972f851b604138ffea43ff5512d7a5ffe18c98be460L351

Or get it from https://github.com/barryhunter/u-lora/tree/patch-1

Which was found when comparing with https://github.com/epeters13/pyLoraRFM9x/
(so they get the credit for getting it working!)

Can also use `receive_all` on the receiver with current u-lora library.