martynwheeler/u-lora #15: Continue refactor and testing of the code

rkompass, 2023-10-25:
@davefes: I continue here from [our micropython discussion](https://github.com/orgs/micropython/discussions/12772):

I got the module connected. No SPI problems. Had to correct my code again (still little flaws:)

`lora_rfm95.py`:
```py
from ucollections import namedtuple
from urandom import getrandbits
from umachine import Pin
from utime import time, sleep_ms

# -----  Constants -----
MODE_SLEEP  = const(0x00)
MODE_STDBY  = const(0x01)
MODE_TX     = const(0x03)
MODE_RXCONT = const(0x05)
MODE_CAD    = const(0x07)
LONG_RANGE_MODE = const(0x80)

class LoRa(object):

    Bw125Cr45Sf128 = (0x72, 0x74, 0x04)   # Bw = 125 kHz, Cr = 4/5, Sf = 128chips/symbol, CRC on. Default medium range
    Bw500Cr45Sf128 = (0x92, 0x74, 0x04)   # Bw = 500 kHz, Cr = 4/5, Sf = 128chips/symbol, CRC on. Fast+short range
    Bw31_25Cr48Sf512 = (0x48, 0x94, 0x04) # Bw = 31.25 kHz, Cr = 4/8, Sf = 512chips/symbol, CRC on. Slow+long range
    Bw125Cr48Sf4096 = (0x78, 0xc4, 0x0c)  # Bw = 125 kHz, Cr = 4/8, Sf = 4096chips/symbol, low data rate, CRC on. Slow+long range
    Bw125Cr45Sf2048 = (0x72, 0xb4, 0x04)  # Bw = 125 kHz, Cr = 4/5, Sf = 2048chips/symbol, CRC on. Slow+long range

    def __init__(self, spi, cs, pirq, addr, res=None, freq=868.0, tx_power=14,
                 modem_config=Bw125Cr45Sf128, rcv_all=False, acks=False, crypto=None):
        """
        LoRa(spi, cs, itr, res=None, addr, freq=868.0, tx_power=14,
                 modem_conf=Bw125Cr45Sf128, receive_all=False, acks=False, crypto=None)
        spi: spi object
        cs: chip select (NSS) pin (mode = Pin.OUT, value=1)
        pirq: interrupt pin (mode = Pin.IN)
        res: reset pin (optional, mode = Pin.OUT, value=1)
        addr: address of this device [0-254]
        freq: frequency in MHz (default: 868.0)
        tx_power: transmit power in dBm (default: 14)
        modem_config: modem configuration, (default: LoRa.Bw125Cr45Sf128 is compatible with the Radiohead library)
        rcv_all: if True, don't filter packets on address
        acks: if True, request acknowledgments
        crypto: if desired, an instance of ucrypto AES (https://docs.pycom.io/firmwareapi/micropython/ucrypto/) - not tested
        """
        
        self.spi = spi              # spi object
        cs(1); self.cs = cs         # chip select (NSS) pin
        self.pirq = pirq            # interrupt pin
        self.res = res              # optional reset pin

        self.addr = addr            # other arguments
        self.freq = freq
        self.tx_power = max(min(tx_power, 23), 5)
        self.modem_config = modem_config
        self.rcv_all = rcv_all
        self.acks = acks
        self.crypto = crypto

        self.mode = None
        self._last_header_id = 0
        self._last_payload = None
        self._cad = None
        self.cad_timeout = 0
        self.send_retries = 2
        self.wait_packet_sent_timeout = 0.2
        self.retry_timeout = 0.2
        
        pirq.irq(trigger=Pin.IRQ_RISING, handler=self._handle_interrupt)   # setup interrupt

        self.reset()                                                       # reset the board
[... 241 lines cut ...]
spi = SPI(2, SPI.MASTER, baudrate=500_000, polarity=0, phase=0) # SCK=PB13, MISO=PB14, MOSI=PB15

# WeAct Blackpill STM32F411: We use SoftSPI
# from umachine import SoftSPI
# spi = SoftSPI(baudrate=100_000, polarity=0, phase=0, sck=Pin('PB13'), mosi=Pin('PB15'), miso=Pin('PB14'))

cs=Pin('PB12', Pin.OUT, value=1)
pirq=Pin('PA8', Pin.IN)
res=Pin('PA9', Pin.OUT, value=1)

led=Pin('PC13', Pin.OUT, value=0)

lora = LoRa(spi, cs=cs, pirq=pirq, res=res, addr=CLIENT_ADDRESS, modem_config=LoRa.Bw125Cr45Sf128, tx_power=LORA_POW)

# loop and send data
#while True:
#    lora.send_to_wait('Test msg', SERVER_ADDRESS)
lora.send('Test msg', SERVER_ADDRESS)
print ('sent')
sleep(1)
led(1)    # led off
sleep(0.5)
```

As it seems I have to solder another module (got 5) for receiving.

davefes, 2023-10-25:
Good to re-sync files.  Receiver still not working, I will check that I am getting interrupts from the RFM96W.

davefes, 2023-10-25:
`if self._freq >= 779:` to `if self.freq >= 779:`

davefes, 2023-10-25:
No interrupts coming from DIO0 to ('B0')

davefes, 2023-10-25:
Getting interrupts from the RFM96W.  The callback doesn't seem to be triggered.

davefes, 2023-10-25:
Possibly in this section:
```py

            if packet_len >= 4:
                header_to = packet[0]
                header_from = packet[1]
                header_id = packet[2]
                header_flags = packet[3]
                message = bytes(packet[4:]) if packet_len > 4 else b''

                if (self._this_address != header_to) and ((header_to != BROADCAST_ADDRESS) or (self._receive_all is False)):
                    return

                if self.crypto and len(message) % 16 == 0:
                    message = self._decrypt(message)

                if self._acks and header_to == self._this_address and not header_flags & 0x80:   # FLAGS_ACK = 0x80
                    self.send_ack(header_from, header_id)

                self.set_mode(MODE_RXCONT)

                self._last_payload = namedtuple("Payload",
                    ['message', 'header_to', 'header_from', 'header_id', 'header_flags', 'rssi', 'snr']
                )(message, header_to, header_from, header_id, header_flags, rssi, snr)

                if not header_flags & FLAGS_ACK:
                    self.on_recv(self._last_payload)
```

davefes, 2023-10-25:
I see logic changes on lines 273 and 279, so over to you!

rkompass, 2023-10-25:
```
From: 1
Received: b'Test msg'
RSSI: -77.0; SNR: 9.75
From: 1
Received: b'Test msg'
RSSI: -77.0; SNR: 10.25
From: 1
Received: b'Test msg'
RSSI: -75.93; SNR: 10.25
From: 1
Received: b'Test msg'
RSSI: -75.93; SNR: 10.0
.......
```
with
```py
from utime import sleep
from lora_rfm95 import LoRa
from umachine import Pin

MODE_RXCONT = const(0x05)

# Lora Parameters
LORA_POW = 12
CLIENT_ADDRESS = 1
SERVER_ADDRESS = 2

# This is our callback function that runs when a message is received
def on_recv(payload):
    print("From:", payload.header_from)
    print("Received:", payload.message)
    print("RSSI: {}; SNR: {}".format(payload.rssi, payload.snr))

# WeAct Blackpill STM32F411: We use SPI2 with (NSS, SCK, MISO, MOSI) = (Y5, Y6, Y7, Y8) = (PB12, PB13, PB14, PB15)
from pyb import SPI
spi = SPI(2, SPI.MASTER, baudrate=500_000, polarity=0, phase=0) # SCK=PB13, MISO=PB14, MOSI=PB15

# WeAct Blackpill STM32F411: We use SoftSPI
# from umachine import SoftSPI
# spi = SoftSPI(baudrate=100_000, polarity=0, phase=0, sck=Pin('PB13'), mosi=Pin('PB15'), miso=Pin('PB14'))

cs=Pin('PB12', Pin.OUT, value=1)
pirq=Pin('PA8', Pin.IN)
res=Pin('PA9', Pin.OUT, value=1)

led=Pin('PC13', Pin.OUT, value=0)

lora = LoRa(spi, cs=cs, pirq=pirq, res=res, addr=SERVER_ADDRESS, modem_config=LoRa.Bw125Cr45Sf128, tx_power=LORA_POW)

# set callback
lora.on_recv = on_recv

# set to listen continuously
lora.set_mode(MODE_RXCONT)

# loop and wait for data
while True:
    sleep(0.1)
```
and not only the `_freq` had to be replaced by `freq`, also the `_acks` by `acks`.

rkompass, 2023-10-25:
If the client script has
```py
while True:
    led(1)    # led off
    lora.send('Test msg', SERVER_ADDRESS)
    # lora.send_to_wait("This is a test message", SERVER_ADDRESS)
    print("sent")
    sleep(1)
    led(0)    # led off
    sleep(1)
```
in the loop then I get the received message every time the message is sent.
If I change the comments to `lora.send_to_wait("This is a test message", SERVER_ADDRESS)` activated
I get the received message only every second time the message is sent. That seems strange.

davefes, 2023-10-25:
changed `self._acks' to `self.acks` and still not go.

Every second time ... maybe as per Barry Hunter's comments change to `retry_timeout` say 0.5 to 1 second.

davefes, 2023-10-25:
Is normal operation for it to return?
```py
                if (self._this_address != header_to) and ((header_to != BROADCAST_ADDRESS) or (self._receive_all is False)):
                    return
```

davefes, 2023-10-25:
With the original `retry_timeout' it will take about 1 second to do a `send_to_wait().  There was a discussion about this at:
https://github.com/martynwheeler/u-lora/pull/13

rkompass, 2023-10-25:
I put the driver and server and client file here:
[lora_rfm95.zip](https://github.com/martynwheeler/u-lora/files/13170959/lora_rfm95.zip)

Note some cosmetic: Now the modes
```py
    # Operating modes
    MODE_SLEEP  = const(0x00)
    MODE_STDBY  = const(0x01)
    MODE_TX     = const(0x03)
    MODE_RXCONT = const(0x05)
    MODE_CAD    = const(0x07)
    LONG_RANGE_MODE = const(0x80)
```
may be used from outside as in` lora.set_mode(LoRa.MODE_RXCONT)    # set to listen continuously`.

Hope you have success with that.

davefes, 2023-10-25:
One coding question, what does:
`cs(1); self.cs=cs`
mean?

rkompass, 2023-10-25:
These are two instructions (may be on two lines without the separating `;`).
The first: `cs(1)` is a shorter and faster version of `cs.value(1)`. The second is the storage of the cs pin object as a class instance variable. As long as we are in `__init__()` we have cs as argument and may change the level to high, i.e. 1 by using `cs(1)`. In other methods access to cs is only possible through the class instance itself, so we use `self.cs(1)` or `self.cs(0)`.

davefes, 2023-10-25:
Ah, `cs.value(1)`  Thank you.

I had to comment-out the assert for the transmitter to work.

rkompass, 2023-10-25:
[image]

rkompass, 2023-10-25:
> I had to comment-out the assert for the transmitter to work.

Should not be necessary any longer.

I feel I have to start learning about LoRa by now ...

davefes, 2023-10-25:
O dear, had to comment-out the assert in the receiver and also not seeing any interrupts.

rkompass, 2023-10-25:
In line 260 `self._receive_all` still is to be replaced by `self.rcv_all`:
`if (self.addr != header_to) and (header_to != 255) and (self.rcv_all is False):  # 255 = BROADCAST_ADDRESS`

If you have to comment out the assert to avoid the "LoRa initialization failed" message, then there is a SPI communication problem. That may also be responsible for the receiving not happening. ???

davefes, 2023-10-25:
Sorry, I need to modify lora_rfm95.py to 433MHz.  I am barely keeping-up!

rkompass, 2023-10-25:
No problem, but how about the SPI issue? That will not change.
I would try to solve this first. Did you try with the latest driver version above? (I sometimes forget to upload the changed driver to the board).

davefes, 2023-10-25:
I needed to re-assign the pins in server and client to what I was actually using!

Also, changed to SoftSPI and "wahoo" the signal appears at the receiver!!!

I believe I have the latest version out of the .zip file.  I must get rid of the hundreds of unwanted "ulora" file around.

Assert statement back-in and all seems good.  Stress-testing underway while I relax!

Because of strange SNR readings at low signal levels I put this in the server:
```py
def on_recv(payload):
    if payload.rssi < -120:
        print('RSSI is below -120dBm')
    else:
        print('From:', payload.header_from)
        print('Received:', payload.message)
        print('RSSI: {}; SNR: {}'.format(payload.rssi, payload.snr))
```

Did you read through https://github.com/micropython/micropython-lib/tree/master/micropython/lora

davefes, 2023-10-25:
I still think that there is some critical timing issue before the `assert` statement.

rkompass, 2023-10-25:
Congratulations. I will have to go to sleep now.

davefes, 2023-10-25:
For 800MHz I would use a 1/10 or 1/4 watt resistor with leads about 1-2mm long.

rkompass, 2023-10-25:
We can start testing more functionality tomorrow. I have the feeling the requesting acknowledgements does not work.
Broadcasting has to be tested too.
There should be an init() function, with more settings.

rkompass, 2023-10-26:
If `send_to_wait()` is used the (listening) server is requested to send an acknowledgement for every packet. This does not work correctly.
The first such acknowledgement sent always reports a timeout. `send_to_wait()` has to retry again each time.

I guess there is an error due to it happening from within the RX interrupt callback while needing another TX interrupt.
Couldn't figure that out more precisely yet.

If you want to investigate: here are the files with many debugging print statements.
[Lora.zip](https://github.com/martynwheeler/u-lora/files/13182267/Lora.zip)

davefes, 2023-10-26:
I noticed both of his examples had `acks=True` so I thought that it must be working.

The comment about longer `retry_timeout` I gather refer to the PICO.  I don't know what timing precision that the STM32F411 has.

My testing with sf=2048 has been rock-solid.  Unfortunately, I am left wondering why your code runs on both platforms and yest the original ulora failed on one platform.

I will have a look but interrupts are not one of my strengths.

rkompass, 2023-10-26:
> I am left wondering why your code runs on both platforms and yest the original ulora failed on one platform.

This is because the SPI configuration is outside the class/module as it should be.
The original ulora failed only because of that.
Now, the error I described only becomes apparent with the debugging by the print statements. Sending everything twice otherwise is not noted because the server in the interrupt routine stays at a point where it sends the acknowledgement back.

davefes, 2023-10-26:
I only have one USB cable that will plug into a WeAct dev board :(  So, I can't run two terminal programs.  Will try to make-up a cable.

Can you explain why SPI fails when it is in the class?

Is the rule:  send() with no acks and send_to_wait() with acks.  No other combinations possible?

davefes, 2023-10-26:
Got it connected, modified pin assignments and have bothe RX and TX running.  Things appear to be working to the untrained eye.

I see acks sometimes on the first send_to_wait() but mostly on the 2nd send_to_wait().   I see a lot of  `no cad detected'

davefes, 2023-10-26:
`
sent 112
int_cad_done: no cad detected
send_to_wait attempt 1
int_tx_done
int_cad_done: no cad detected
send_to_wait attempt 2
int_tx_done
int_rxcont packet id:113 from:2 to:1 req_ack:False ack:True
got ack
`

davefes, 2023-10-26:
An observation ... I forgot to change to 433MHz in lora_rfm95.py and the transmitter initialises and appears to send data to who knows where.  Maybe, having `freq` in sever/client scripts might be useful.

rkompass, 2023-10-26:
Same here.
I have to rewrite all uses of `time.time()` because that has seconds resolution to use of time.ticks_ms() and time.ticks.diff().
Because the timeouts are defined in fractions of seconds.
Tomorrow.

davefes, 2023-10-27:
Just an idea to test with:
make both 'retry_timeout` and `wait_packet_sent_timeout' =1

And in client maybe increase the time between sends:
```py
while True:
    led(0)    # led on
    
    # Here we send just a test message. Packets ar>
    # lora.send('Test message {}'.format(i), SERVE>
    
    # Here we send a test message with request for>
    lora.send_to_wait('Test message with ack {}'.f>
    
    print('sent {}'.format(i))
    i += 1
    sleep(1)
    led(1)    # led off
    sleep(4)  # to allow for 4 retries
```
In testing with Wei's LoRa code https://github.com/Wei1234c/SX127x_driver_for_MicroPython_on_ESP8266/tree/master I found that I had to put say a 1 second delay between sends to allow the receiver to process the message.  Maybe, that is what the whole retry timeout thing is for.

Good luck.

davefes, 2023-10-27:
Updated a comment on the original thread about using a resistor for the antenna load.

What I showed in the picture is barely appropriate for 433MHz,  Suggest a 47 to 56Ohm 0805 SMD resistor soldered between the ANT and adjacent ground pin.

rkompass, 2023-10-27:
This [video](https://www.youtube.com/watch?v=U8__pLUnfwA) shows what I will probabably try in the end.

davefes, 2023-10-27:
That is one way.  I cut off about 5-6mm of the outer telfon off, flow solder on the braid, then cut off say 3mm of braid, wrap thin copper wire around the braid (instead of making pig-tails), solder it, leave 2-3mm of wire sticking out to solder to ground and then cut 1-2mm of the inner Teflon off to expose the inner connector. 

I have been running send_to_wait() most of day and seems to work for me.  I have 1 seconds timeouts and using Sf=2048.

If you are sending a long message at a high Sf do you not need longer timeouts?

> Can_ you explain why SPI fails when it is in the class?

> Is the rule: send() with no acks and send_to_wait() with acks. No other combinations possible?

davefes, 2023-10-27:
If you give me a screenshot of the unwanted behaviour that you are seeing it would help to focus my attention.  Testing down near the -120dBm level with `send_to_wait()` I only see correct messages.  When I was testing just with `send()` I would sometimes get corrupted messages.

As per question 2) acks were off.  As CRC is on why does the receiver display a corrupted message?  If the software, when using `send_to_wait()` can tell the transmitter to send the message again, why can't the receiver processing determine whether or not to display a corrupted message?

My typical use case is  sending brief messages, say once an hour and over long distances.  Power source 1 * 18650 cell and 1W PV panel.  1-4seconds every hour with the STM32F411 going to deepsleep seems a good solution.

davefes, 2023-10-27:
My search attempts for how interrupts are used with the RFM9x.  I can't find RadioHead RH_RF95.c on his site but this:
https://github.com/PaulStoffregen/RadioHead/blob/master/RH_RF95.cpp
Comments in `void RH_RF95::handleInterrupt()`

Also Wei makes this comment in his code:

>     # on RPi, interrupt callback is threaded and racing with main thread,
    # Needs a lock for accessing FIFO.
    # https://sourceforge.net/p/raspberry-gpio-python/wiki/Inputs/
    # http://raspi.tv/2013/how-to-use-interrupts-with-python-on-the-raspberry-pi-and-rpi-gpio-part-2

Good luck!

davefes, 2023-10-27:
Just another thought ... if you place a probe on DIO0 using a good DSO do you see multiple interrupts during 1 send_to_wait() call?  

Here is a picture using a DSO138:
[image]
... with send() you don't get two rising edges, ie only the 5ms pulse.

davefes, 2023-10-28:
There is a lot of processing in `_handle_int`, (9 SPI read/write calls) maybe increasing the SPI clock rate.

rkompass, 2023-10-28:
Tomorrow I will give a description. Also my feeling is now that systematic testing is needed.

All this is about `send_to_wait()`. The receiver switches to sending the requested ACK which is now scheduled, i.e. happens outside the interrupt routine (which is fine). In sending the ACK it does a CAD (carrier detection) like always before before sending (we do not want to send into a just proceeding transmission). But this CAD is not finished (no CAD_DONE interrupt). Instead the CAD times out and transmit is started.
We see a report of the CAD_DONE interrupt in a TX Mode however, which is reported, as I have included a print on any interrupt happening.
You would from this suppose that the CAD phase was not long enough, but setting the cad_timeout longer seems to make the problem not go away. This has to be investigated.

Here is the new version, also with little cosmetics/renaming. Hope you get a grip on the different reports and the problem.
I may go into deeper dialogue about this tomorrow.
[Lora.zip](https://github.com/martynwheeler/u-lora/files/13196137/Lora.zip)

davefes, 2023-10-28:
Wow, I can see you have put a lot of work into this.

Are you wanting to use send_to_wait () to get higher message reliability?  I have managed with just testing the received message at the server, even for messages with small value changes in them.

Here is what I see:

Client:
`
set CAD mode
int_cad_done: no cad detected
send_to_wait attempt 1
int_tx_done
int_rxcont packet id:165 from:2 to:1 req_ack:False ack:True
got ack
sent 165
`
Server:
`
other interrupt: mode: 3, irq_flags: 0x4
int_tx_done
int_rxcont packet id:165 from:1 to:2 req_ack:True ack:False
send ack to 1 at id 165
set CAD mode
wait_cad: timeout
From 1:
Received: b'Test message with ack 165'
RSSI: -105.33; SNR: 8.25
`
So, the problem is the `no cad detected` on the client`, correct?

I am off to the "Big city" for a few days so things will be quiet from my end.

davefes, 2023-10-28:
Is the real problem:
`wait_cad: timeout`
at the server?

davefes, 2023-10-29:
Also, the server setting CAD mode after sending an ACK doesn't seem right.

davefes, 2023-10-29:
https://www.thethingsnetwork.org/forum/t/channel-activity-detection-cad-implementation-of-lora/13373

rkompass, 2023-10-29:
Thanks, this may be the solution. Although in principle this should be possible with WAIT_CAD too.--Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

davefes, 2023-10-29:
My current understanding ... a point to point link returning an `ack` to the client has good value.  Maybe, in complex systems and/or wanting to save receiver current in a duplex system, CAD may have some value.

rkompass, 2023-10-29:
I just tried the micropython-lib SX1276 version of reliable sending and receiving. Got sender and receiver to run, but there was no communication. I will have to repeat this with the SX1276 boards that I got yesterday (another run of soldering).
Noted that in Angus Grattons code there are a few very interesting comments.
He put a delay of 100ms (at least 25 ms) for sending the ACK. Perhaps this is the culprit? It could be that the sender is not able to switch back fast enough from sending to receiving again.
I like the current code we have - compared to the micropython-lib one it is quite compact. But there is quite some work to be done yet. It takes time to investigate things thoroughly. I would like to have a decent introduction with code examples for the RFM95 but didn't find some.
------
Did more modifications and testing.
The thing is that with switching from TXCONT to doing a CAD the CAD times out and the CAD_DONE interrupt is noted when DR mode is already active.
This seems to be independent from a delay inserted in between TXCONT and SLEEP and CAD.
So the above assumption does not hold.
Sending the ACK alone seems to work, then the CAD_DONE is correctly noted.
I cannot explain this atm. Perhaps an interrupt issue.  It needs experimentation.
All nested interrupt stuff is gone now by usage of `micropython.schedule` - this is the correct way to do it.

davefes, 2023-10-30:
I found when testing Wei's send/receive that I had to put a short delay after a tx. I think 250ms was OK, but I put in 1 second delays.

davefes, 2023-11-01:
Just an observation ... `Bw125Cr45Sf2048` seems to work well, but with `Bw125Cr48Sf4096` the message comes through 4 times.

davefes, 2023-11-12:
@rkompass The main issue I am facing is that my sensor links using Wei's code may send a corrupted message, but at least an hour later it tries again ... so the odd missed data is not a problem.  On a LoRa control link "missed messages" are a problem.

In `ulora.py` , the `send_wait()` method tries 4 times and the real plus is that you get an ACK sent back to the sender.

Have you made any further progress?

rkompass, 2023-11-18:
> I just tried the micropython-lib SX1276 version of reliable sending and receiving. Got sender and receiver to run, but there was no communication. I will have to repeat this with the SX1276 boards that I got yesterday (another run of soldering).

@davefes I saw that you posted the same observation on MP discussions. Cannot solve it at the moment, but will solve it later.

davefes, 2023-11-18:
Appears that any "PA  switching " that takes place is turning the whole TX on/off, as well as any possible antenna port selection.